MajicDesigns/MD_UISwitch #2: False positive for KEY_SINGLE

Nagidal, 2018-01-01:
Hello Marco, 

I have tried your UI_Switch library with the test code below and I'm getting a false positive KEY_SINGLE detect about 1 in 20 times or even more often when I press the standard Arduino push button (attached image borrowed from ladyada.net)
[image]

I used the following code (also attached):
```
#include <MD_UISwitch.h>

// Define what type of testing is being done
#define TEST_DIGITAL_SIMPLE 1

#if TEST_DIGITAL_SIMPLE
#define TITLE "Simple Digital"

const uint8_t DIGITAL_SWITCH_PIN = 2;       // switch connected to this pin
const uint8_t DIGITAL_SWITCH_ACTIVE = LOW;  // digital signal when switch is pressed 'on'

MD_UISwitch_Digital S(DIGITAL_SWITCH_PIN, DIGITAL_SWITCH_ACTIVE);
#endif

void setup() {
  Serial.begin(57600);
  Serial.print(F("\n[MD_UISwitch "));
  Serial.print(F(TITLE));
  Serial.print(F(" Example]"));

  S.begin();
  S.enableDoublePress(false);
  S.enableLongPress(false);
  S.enableRepeat(false);
  S.enableRepeatResult(true);

}

void loop() {
  // put your main code here, to run repeatedly:
  MD_UISwitch::keyResult_t k = S.read();

  switch(k)
  {
    case MD_UISwitch::KEY_NULL:      /* Serial.println("NULL"); */  break;
    case MD_UISwitch::KEY_PRESS:     Serial.print("\nKEY_SINGLE "); break;
    case MD_UISwitch::KEY_DPRESS:    Serial.print("\nKEY_DOUBLE "); break;
    case MD_UISwitch::KEY_LONGPRESS: Serial.print("\nKEY_LONG   "); break;
    case MD_UISwitch::KEY_RPTPRESS:  Serial.print("\nKEY_REPEAT "); break;
    default:                         Serial.print("\nKEY_UNKNWN "); break;
  }
  if (k != MD_UISwitch::KEY_NULL)
  {
    if (S.getKey() >= ' ')
    {
      Serial.print((char)S.getKey());
      Serial.print(" ");
    }
    Serial.print("[0x");
    Serial.print(S.getKey(), HEX);
    Serial.print("]");
  }

}
```
[Debounce_test.zip](https://github.com/MajicDesigns/MD_UISwitch/files/1596277/Debounce_test.zip)

MajicDesigns, 2018-01-01:
The switch you show is called a 'tact' switch and is a momentary-on keyswitch.

Explain a 'false positive KEY_SINGLE' means as there is no explanation of what you expect to see, what you are seeing and why that should be considered a problem?

Nagidal, 2018-01-06:
Well, most of the time, when I press the button once, the serial console writes a singe event KEY_SINGLE. But sometimes, about 1 in 20 times or more often I get two such events after just one press.

MajicDesigns, 2018-01-06:
OK. It does not happen for me so if you find specific reasons for it let me know.

closed 2018-02-11 by MajicDesigns