MajicDesigns/MD_UISwitch #4: Example MD_UISwitch compilation error

jirmjos, 2018-02-11:
Good day.

I need some help to solve a extrange issue.

I have a compilation error seems because some issue with the library code, when try to compile example sketch for tree analog switch buttons attached on a ESP8266 Wemos D1 mini:

"Arduino\libraries\MD_UISwitch\src\MD_UISwitch.cpp:43:5: error: return-statement with no value, in function returning 'MD_UISwitch::keyResult_t' [-fpermissive]"

Below my code (parsed from example to test only analog switch readings), but I tried too the example code downloaded with the library with same result:

`#include <MD_UISwitch.h>

#define TITLE "Analaog Input"

const uint8_t ANALOG_SWITCH_PIN = A0;       // switches connected to this pin

// These key values work for most LCD shields
MD_UISwitch_Analog::uiAnalogKeys_t kt[] =
{
  { 672, 15, 'R' },  // Right
  { 274, 15, 'C' },  // Left
  { 145, 15, 'L' },  // Down
  { 945, 15, 'D' },  // Right+Center
  { 405, 15, 'I' },  // LEft+Center
};

MD_UISwitch_Analog S(ANALOG_SWITCH_PIN, kt, ARRAY_SIZE(kt));

void setup(void)
{
  Serial.begin(115200);
  Serial.print(F("\n[MD_UISwitch "));
  Serial.print(F(TITLE));
  Serial.print(F(" Example]"));

  S.begin();
  S.enableDoublePress(false);
  S.enableLongPress(false);
  //S.enableRepeat(false);
  S.enableRepeatResult(true);
}

void loop()
{
  MD_UISwitch::keyResult_t k = S.read();

  switch (k)
  {
    case MD_UISwitch::KEY_NULL:      /* Serial.println("NULL"); */  break;
    case MD_UISwitch::KEY_PRESS:     Serial.print("\nKEY_SINGLE "); break;
    case MD_UISwitch::KEY_DPRESS:    Serial.print("\nKEY_DOUBLE "); break;
    case MD_UISwitch::KEY_LONGPRESS: Serial.print("\nKEY_LONG   "); break;
    case MD_UISwitch::KEY_RPTPRESS:  Serial.print("\nKEY_REPEAT "); break;
    default:                         Serial.print("\nKEY_UNKNWN "); break;
  }
  if (k != MD_UISwitch::KEY_NULL)
  {
    if (S.getKey() >= ' ')
    {
      Serial.print((char)S.getKey());
      Serial.print(" ");
    }
    Serial.print("[0x");
    Serial.print(S.getKey(), HEX);
    Serial.print("]");
  }
}`

MajicDesigns, 2018-02-11:
Fixed and repo updated. Version 1.2.1

closed 2018-02-11 by MajicDesigns

jirmjos, 2018-02-11:
Working now.

Thank you Marco.