MajicDesigns/MD_UISwitch #8: HowTo detect 2 keys pressed simultaneously with MD_UISwitch_Digital()

wolfgamM, 2019-02-23:
Hello,

pressing 2 keys simultaneously is often used to enter some kind of setup or factory default procedures.
I am facing the problem to detect 2 simultaneously pressed keys with MD_UISwitch_Digital().

Pressing one button for long time results in 'key repeat' answers from .read method.
When pressing the second button, the .read method answers 'KEY_NULL' as if no button is pressed.

regards
Wolfgang

MajicDesigns, 2019-02-23:
Use 2 instances of the UISwitch objects to detect 2 keys at the same time.

You use case is interesting and not really well handles by UISwitch - I'll think about what can be done to support this better in a future release.

wolfgamM, 2019-02-25:
Hello,

>Use 2 instances of the UISwitch objects to detect 2 keys at the same time.
With 2 instances you run into the problem that keyResult_t enumeration is not sufficiant.
KEY_RELEASED, KEY_CLICKED and KEY_DBLCLICKED would be helpful.

Assume:
1)  pressing 2 keys cannot be done simultaneously by a user. There is always a time difference.
2)  normal user operation is startet with: KEY_PRESS

Problem:
With 2 instances, you need 2 subsequent .read() methods which are delayed at least for the debounce time. With getting the first KEY_PRESS (after debounce time), you cannot know whether the second key comes soon or not. So a normal user action starts with KEY_PRESS (after debounce time).

Solution:
a) A possible solution is to check for actions on an alternative port during debounce time
b) to introduce new members to keyResult_t enumeration.
User Action starts alternatively on a KEY_CLICKED. 
When using KEY_PRESSED there can be several checks on other KEY_PRESSED instances with KEY_RELEASED.

regards
Wolfgang

MajicDesigns, 2019-02-25:
The library needs a KEY_DOWN and a KEY_UP notification for what you want to do. This means the way debouncing is done in the library needs to change as well as the current structure does not support this easily.
As mentioned in the first response, I'll probably add this in a future release but I have no plan for this yet. I would also not expect that the 'two key' solution will be implemented in the library as this is too specific, but the additional notifications will support the user code to do what it needs.

MajicDesigns, 2019-03-12:
Version 2.0.0 implements KEY_DOWN and KEY_UP events notification.

closed 2019-03-12 by MajicDesigns