MajicDesigns/MD_UISwitch #9: Version 2.0 is not backward compatible with 1.2.x

alf45tar, 2019-03-17:
I using your fantastic library for my PedalinoMini project but with the latest version 2.0 nothing is working ad before. Most of the time press is not reported or block the execution.
PedalinoMini is working on ESP32 board with Arduino framework.

Thanks
alf45tar

MajicDesigns, 2019-03-17:
I have tested the library before release and, aside from the KEY_DOWN and KEY_UP that are new, the notifications of the other key states is substantially identical to the previous version.

Can you please try the  UI_Switch_Example code and test whether you see the same non function with with that code - it is very simplified code that just simply prints out what key type has been pressed.

alf45tar, 2019-03-18:
UI_Switch_Example works like a charm
I will check better my code

MajicDesigns, 2019-03-18:
Ok, Let me know if you need more info.

closed 2019-03-18 by MajicDesigns