MajicDesigns/MD_UISwitch #10: Issue in 2.0.0 with multiple MD_UISwitch_Digital instances

dotorg, 2019-04-02:
## Subject of the issue
I'm having an issue with the library where it does't seem to handle having multiple instances of the MD_UISwitch_Digital class instantiated in the code. Not sure if I have something configured wrong, but based on the documentation, this seems correct. (See below.)

## Your Environment
**Library Version:** 2.0.0
**Arduino IDE version:** 1.8.8
**Hardware model/type:** ESP8266 / Wemos D1

## Steps to Reproduce
With the code below, if you don't define MDUISWITCH_ENABLE and log the raw button states, the code outputs correctly.  (In my case B1 and B2 have external pull-downs, B3 uses an internal pull-up, but the issue presents itself regardless of which buttons are wired up where...)

When using MD_UISwitch_Digital, BTN2 and BTN3 do not respond to any changes on the GPIO pins. For example, a single press of BTN1 yields:
B3: KEY_DOWN
B1: KEY_DOWN
B3: KEY_UP
B1: KEY_UP
B3: KEY_PRESS
B1: KEY_PRESS

BTN2 doesn't respond at all, although it does work correctly without MD_UISwitch.

BT3 single push generates:
B2: KEY_DOWN
B3: KEY_DOWN
B1: KEY_DOWN
B2: KEY_UP
B3: KEY_UP
B1: KEY_UP
B2: KEY_PRESS
B3: KEY_PRESS
B1: KEY_PRESS

If I downgrade to 1.2.1, things work fine, so this is something new in 2.0.

## Code Demonstrating the Issue
````
#include <MD_UISwitch.h>

#define MDUISWITCH_ENABLE

#ifdef MDUISWITCH_ENABLE

MD_UISwitch_Digital BTN1(D1, HIGH);
MD_UISwitch_Digital BTN2(D2, HIGH);
MD_UISwitch_Digital BTN3(D7, LOW);

#endif

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  pinMode(D1, INPUT);
  pinMode(D2, INPUT);
  pinMode(D7, INPUT_PULLUP);

#ifdef MDUISWITCH_ENABLE
  BTN1.begin();
  BTN2.begin();
  BTN3.begin();
#endif
}

void loop() {
 #ifndef MDUISWITCH_ENABLE
  bool b1State = digitalRead(D1);
  bool b2State = digitalRead(D2);
  bool b3State = digitalRead(D7);
  Serial.println("B1: " + String(b1State) + ", B2: " + String(b2State) + ", B3: " + String(b3State));
#endif

#ifdef MDUISWITCH_ENABLE
  MD_UISwitch::keyResult_t b1_result = BTN1.read();
  logButtonState("B1", b1_result);
  MD_UISwitch::keyResult_t b2_result = BTN2.read();
  logButtonState("B2", b2_result);
  MD_UISwitch::keyResult_t b3_result = BTN3.read();
  logButtonState("B3", b3_result);
#endif

  delay(100);
  
}

void logButtonState(const char* buttonName, MD_UISwitch::keyResult_t state)
{
  switch (state)
  {
  case MD_UISwitch::KEY_NULL:      /* Serial.print("KEY_NULL"); */  break;
  case MD_UISwitch::KEY_UP:        Serial.printf("%s: KEY_UP\n", buttonName);     break;
  case MD_UISwitch::KEY_DOWN:       Serial.printf("%s: KEY_DOWN\n", buttonName);     break;
  case MD_UISwitch::KEY_PRESS:      Serial.printf("%s: KEY_PRESS\n", buttonName);     break;
  case MD_UISwitch::KEY_DPRESS:    Serial.printf("%s: KEY_DPRESS\n", buttonName);     break;
  case MD_UISwitch::KEY_LONGPRESS: Serial.printf("%s: KEY_LONGPRESS\n", buttonName);     break;
  case MD_UISwitch::KEY_RPTPRESS:  Serial.printf("%s: KEY_RPTPRESS\n", buttonName);     break;
  default:                         Serial.printf("%s: KEY_UNKNWN\n", buttonName);     break;
  }
}
````

MajicDesigns, 2019-04-06:
Thanks for demonstrating this issue so clearly. The problem is caused by static declarations in the code that should have been changed after standalone testing was completed.

New version will be issued as 2.1.0. I have testing it working, so please test when you can and close the issue if you find it is working.

MajicDesigns, 2019-04-25:
No further response from OP

closed 2019-04-25 by MajicDesigns