MajicDesigns/MD_UISwitch #11: Tolerance to "too long" long press

bdutta, 2020-01-11:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
YES
[ ] Have you checked this has not already been submitted and/or resolved?
YES
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)
CONSIDERED 

## Subject of the issue
My aim was to use this library to detect button presses for a MIDI foot pedal. I am using an UNO and have wired up 4 momentary switches in the internal pullup mode to 4 digital pins. The sample application works, but I made 2 small change, i.e.:
'''
    BTN[i]->setPressTime(250);
    BTN[i]->setLongPressTime(1200);
'''
which get me close the the more typical button press behaviour seen in such pedals, since otherwise, with default values most typical short presses were also being detected as long presses. However, I notice to behaviours which I'd like to change if possible. One is, that once in every 3-5 long-presses, I get one which is followed up by a "ghost short press". This is quite consistent, and this "ghost short press" seems to happen when I release the button after a long press.

Another observation is that long presses that are perhaps a bit too long, get detected as a short press (for the "press" transition) and another short press (apparently for the "release" transition). Can this behaviour be altered in the way that a long press, could be somewhat longer long-press can be had ?
## Your Environment
**Library Version:** 2.1
**Arduino IDE version:** 1.8.10
**Hardware model/type:** Arduino UNO clone (w/ CH340 chip)
**OS and Version:** Lubuntu 19.10 64-bit on host PC

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your code here.
````

MajicDesigns, 2020-01-11:
Can you please provide minimal code that creates this issue on your system. This behaviour is not seen on my test system.

bdutta, 2020-01-11:
The code is almost same as the example, with some minor changes:

````
// Example showing use of the MD_UISwitch library
// 
// Exercises many individual digital switches, prints the 
// switch value on the Serial Monitor
//
#include <MD_UISwitch.h>

// define pin numbers for individual switches
const uint8_t SW_PIN[] = { 4, 5, 6, 7 };

MD_UISwitch_Digital *BTN[ARRAY_SIZE(SW_PIN)];

void logButtonState(const char* buttonName, MD_UISwitch::keyResult_t state)
{
  if (state == MD_UISwitch::KEY_NULL)
    return;

  //Serial.print("\n");
  Serial.print(buttonName);
  Serial.print(": ");

  switch (state)
  {
  case MD_UISwitch::KEY_NULL:      /*Serial.print("KEY_NULL");*/  break;
  case MD_UISwitch::KEY_UP:        Serial.print("KEY_UP | ");        break;
  case MD_UISwitch::KEY_DOWN:      Serial.print("KEY_DOWN | ");      break;
  case MD_UISwitch::KEY_PRESS:     Serial.println("KEY_PRESS");     break;
  case MD_UISwitch::KEY_DPRESS:    Serial.println("KEY_DPRESS");    break;
  case MD_UISwitch::KEY_LONGPRESS: Serial.println("KEY_LONGPRESS"); break;
  case MD_UISwitch::KEY_RPTPRESS:  Serial.println("KEY_RPTPRESS");  break;
  default:                         Serial.println("KEY_UNKNWN ####");    break;
  }
}

void setup(void)
{
  Serial.begin(9600);

  for (uint8_t i = 0; i < ARRAY_SIZE(BTN); i++)
  {
    BTN[i] = new MD_UISwitch_Digital(SW_PIN[i], LOW);
    BTN[i]->setPressTime(250);
    BTN[i]->setLongPressTime(1200);
    BTN[i]->begin();
  }
  Serial.println("Started...");
}

void loop(void)
{
  for (uint8_t i = 0; i < ARRAY_SIZE(BTN); i++)
  {
    char name[] = "B ";

    name[1] = i + '0';  // turn into a digit
    logButtonState(name, BTN[i]->read());
  }
}

````

MajicDesigns, 2020-01-19:
> One is, that once in every 3-5 long-presses, I get one which is followed up by a "ghost short press".

I am unable to replicate this using your code. The only thing I can suggest is that you may be seeing switch bounce long enough for it to be registered as a key press (this is a long shot). One way to test if the key switch makes a difference is to try it with different switches. FYI I test with tact switches.

> Another observation is that long presses that are perhaps a bit too long, get detected as a short press (for the "press" transition) and another short press (apparently for the "release" [...]

This is probably the key repeat that kicks in when you keep a key pressed after the (long press + key repeat) time period. I suggest you turn off key repeat (BTN[i]->enableRepeat(false)) as this is probably not behaviour you want for this type of application anyway.

closed 2020-01-25 by MajicDesigns