MajicDesigns/MD_UISwitch #13: Thinking about also using PCF8574

Manhem, 2021-08-08:
Hi.
Do you have any plan for using I2C expander PCF8574 or other expanders?
It could be nice if you also can expand you library with some I2C expanders.

It's a very good library to handle buttons.

MajicDesigns, 2021-08-10:
No plans as it would then mean that I need to support X hardware, then Y hardware, etc. and it all quickly gets out of hand. There may be a more general; mechanism that can be made available. I'll look at it sometime soon.

Manhem, 2021-08-11:
Hi.
Thanks.
I hope you can find an easy way to make a general solution.

MajicDesigns, 2021-08-11:
Class MD_UISwitch_User provides generic access that will work with any hardware not directly supported by the library. Implemented in v2.2.0.

Manhem, 2021-08-11:
Thanks.
I shall try this out.

closed 2021-09-06 by MajicDesigns