MajicDesigns/MD_UISwitch #15: Detect multiple concurrent button presses or long presses.

pbinsted, 2022-04-25:
I'm really digging this library, great work.

I am trying to detect when two buttons are held simultaneously for a period of time (longpress).

I've tried a basic IF statement using both switches KEY_DOWN during the loop but have had no luck.

Thanks,
Pete

MajicDesigns, 2022-04-25:
With no code it is hard to say what you are doing wrong.

1. Make sure the switches are separate objects as you need the detection to work concurrently.
2. The status is only sent once when the status is detected. It is unlikely that the detection will be at the same time, so a simple if (switch1==keydown and switch2 == keydown) is highly unlikely to work. You should store the last status non-null status of each switch and compare the two statuses. Issues then arise of how close they need to be for it to register as 'simultaneous' and that needs to be handled by a timer in the application.

pbinsted, 2022-04-26:
Hi. I think you nailed it with point 2. That is what I figured was happening. Thanks so much for the help and the great libraries.

closed 2022-04-26 by pbinsted

MajicDesigns, 2022-04-26:
FWIW I added a new 'Concurrent' example in the latest version (v2.2.1) to test that my idea was correct.